ROBOTIS-GIT/emanual #109: Download links are broken.

tfoote, 2018-02-14:


tfoote, 2018-02-14:
Also the OnShape links https://cad.onshape.com/documents?filter=recently-opened&column=modifiedAt&order=desc&q=turtlebot3 in the open source hardware section. now only shows the TurtleBot3 Tank variant.

routiful, 2018-02-19:
Hello @tfoote  :)

Below links which you point out can be accessed now. 
`http://www.robotis.com/service/download.php?no=690` (OpenManipulator Chain) and 
`http://www.robotis.com/service/download.php?no=678` (TB3 Waffle Pi)

Furthermore, Onshape link is changed. You can show new page as below image
[image]

Best regards
Darby

robotpilot, 2018-02-19:
Hi @tfoote 

There seems to be no way for all users to see the OnShape links you mentioned. The OnShape does not support an advanced search mode (https://forum.onshape.com/discussion/5686/is-it-possible-to-search-the-public-documents-by-owner) for searching by model name and original owner name. Unfortunately, it seems that there is no way to show all original models (just all shared models) as above, but to link each model one by one.

Thank you for your feedback.

tfoote, 2018-02-21:
Thanks the links are working for me now. If they don't have a better search this is probably all we can do.

closed 2018-02-21 by tfoote