ROBOTIS-GIT/emanual #249: More Arduino Information for Pin Interrupts

KurtE, 2018-04-28:
Suggestion: Maybe add some more explicit information about pins that support external Interrupts for Arduino users. 

I am wanting to use an interrupt on a pin from a sensor  and did not find anything when I did a search for Interrupt in the emanual.  So I looked at the variant sources... 

Probably want to mention that, you should use the digitalPinToInterrupt macro like: 
    attachInterrupt(digitalPinToInterrupt(_interrupt_pin), my_isr, FALLING);

Likewise it might be nice if it said that supported pins are:  2, 3, 4, 7, 8, 42, 45, 72

Note: I appreciate that most of this information is already in the emanual if you know where to look for it.   Example in section:
https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/parts/controller/opencr10.md#arduino-connector

It shows some of the pins in Function 3 (but no explanation about what EXTI_0... is. 

Likewise in section: 
https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/parts/controller/opencr10.md#pin-definition

Shows the whole pin definition table, which gives you all the pins, which is great (Thanks!)

Now to see if the interrupts work :D

ROBOTIS-Will, 2018-05-09:
More info will be keep updated.

KurtE, 2018-05-10:
Thanks @ROBOTIS-Will  - The manual is getting better and I appreciate the work you all do!

It is great that you added the information about the IO pins are 5v tolerant.  You might also put in the information about the Analog pins some where. 

Example maybe in the specification section: 
http://emanual.robotis.com/docs/en/parts/controller/opencr10/#specifications
Wondering in that section about Analog: ADC Channels (8bit) x 6
(Is it only 8 bit?)  - My quick look of chapter 15 shows 12,10, 8, 6 bit configurations... 

Might also mention the 5v tolerant info here as well. 

Other suggestion: Maybe add links to the Reference section for where you can download the ST reference PDFs for this processor. 

Minor Typo: Would probably not mention, but noticed it, when looking at Pin information:
Section 3.2 (leds);  Arupinopin 22-25

ROBOTIS-Will, 2018-05-14:
Thank you for the suggestions and typos! 😃
Although I'm going to close this thread as information regarding its title has been updated to the eManual, please don't hesitate to create another thread with your brilliant insight.
Always appreciate for your contribution.

closed 2018-05-14 by ROBOTIS-Will