ROBOTIS-GIT/emanual #302: RViz Commands Missing

arbenton, 2018-05-25:
I've found that the commands for opening the various rviz models are missing from the most recent documentation (found at emanual.robotis.com/docs), and am unable to find them. For instance, the page on slam does not provide the command for opening rviz for mapping. 

They're simple to reconstruct, but its much easier when providing these resources to students to have the relevant rviz commands placed in the documentation right after e.g. starting gmapping. 

Could these be added back in, or at least documented in some way?

robotpilot, 2018-05-30:
Hi @arbenton,

We are avoiding listing too many commands for our customers' convenience.
For the same reason, we modified to execute RViz automatically when xxxxxx.launch is executed.
This change is automatically executed even if the user does not manually execute RViz.

Do you need a description of the separate run?

arbenton, 2018-06-01:
Alright, this makes sense. I'll have my students get the most recent releases. 

It'd be nice for the emanual to have some easily accessible version selection, such as that seen on most software documentation (e.g. the dropdown at the top of https://docs.python.org/3/index.html) - the updates of the past months have been quite disruptive in the classroom.

closed 2018-06-01 by arbenton

ROBOTIS-Will, 2018-06-11:
@arbenton,
As GitHub will automatically generate the file with the latest contents, it's difficult for us to maintain the previous version.
However, you can definitely fork(copy) the eManual version you want to use for your classes, and create your version of eManual site that does not affected by the latest change of our eManual.
Below is the example of repository setting to serves the eManual in your repository.
1. Go to your forked emanual repository.
2. Go to Settings tab.
3. In the Options > Settings section, Set the Repository name as "your_id.github.io"
4. In the Options > GitHub Pages section, set the source as "master branch"
Once you completed above steps, your classroom can use "https://your_id.github.io" as a class material.
Please remember that GitHub allows only one site so if you are already using GitHub site, this will not available.
Thank you.