ROBOTIS-GIT/emanual #354: About Dynamixel API

flilledshark, 2018-06-25:
Excuse me.
I'd like to configure the environment of the API on this page, but it was not supported by visual studio 2017, so it could not be set.
Is there any way to solve it?
The OS you are using is Windows.
http://support.robotis.com/jp/software/dynamixel_sdk/usb2dynamixel/windows/visual_c++.htm

I would like you to tell me.

ROBOTIS-Will, 2018-06-26:
Hello,
The latest contents regarding DynamixelSDK can be found in our new e-Manual.
[http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_windows/#cpp-windows](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_windows/#cpp-windows)
Please do not refer to old manual(support.robotis.com) unless you are using old SDK.
You can also get more help from the issue page of Dynamixel SDK repository below.
[https://github.com/ROBOTIS-GIT/DynamixelSDK/issues](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues)
Thank you.

closed 2018-06-26 by ROBOTIS-Will

flilledshark, 2018-06-26:
It was very helpful.
Thank you very much.