ROBOTIS-GIT/emanual #388: renable actuators after a battery change

DoNck, 2018-07-18:
Hi,

Battery voltage went down to 11V on my Turtlebot 3 so it started beeping and actuators got disabled.
I plugged the power supply, replaced the battery by a charged one and unplugged the power supply.
Now I would like to re-enable the actuators.
Currently the only way I found is a full reboot of the OpenCR + SBC which is pretty cumbersome.

Is there a better way ?

routiful, 2018-07-19:
Hello :)

After you hot swap between battery and power, you need to enable Dynamixels.
Because when voltage has dropped to 11V, firmware make disable to Dynamixels.

So, you are supposed to publish `true` value through topic `/motor_power` after hot swap.
`rostopic pub /motor_power std_msgs/Bool "data: true"` 

By the way, if you have questions about turtlebot3, please use issue page in turtlebot3 repository.

https://github.com/ROBOTIS-GIT/turtlebot3/issues

closed 2018-07-19 by ROBOTIS-Will

routiful, 2018-07-22:
If you need any help, please feel free to open issue :)