ROBOTIS-GIT/emanual #578: Unbale to locate package when installed Additional Dependent Packages for AutoRace on Pi

Ethanaiot, 2018-12-01:
Hi,

 When I installed Additional Dependent Packages for AutoRace on Pi, it has the following issues . Could you please advise how can I fix it?

pi@raspberrypi:~$ sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-proc
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-kinetic-image-transport
E: Unable to locate package ros-kinetic-cv-bridge
E: Unable to locate package ros-kinetic-vision-opencv
E: Unable to locate package ros-kinetic-image-proc

Ethanaiot, 2018-12-04:
Hi Gilbert,
Could you please help? Thank you in advance!!

Regards,
Ethan

ROBOTIS-Will, 2018-12-17:
Hi @Ethanaiot ,
Looks like dependent packages are not installed.
Have you try to reinstall those packages that are not installed properly?

Ethanaiot, 2018-12-22:
Hi Will,  thanks for your reply.

We have terminated the project. Please ignore the issue.
If it is reopened and I still meet the same issue, ill let you know. Thank
you.

Regards,
Ethan

closed 2019-01-15 by ROBOTIS-Will

imostafizur, 2022-07-03:
Please check your ROS version using this command **rosversion -d**.
After that run sudo apt-get install ros-**version_name-**image-transport ros-**version_name**-cv-bridge ros-**version_name-**vision-opencv python-opencv libopencv-dev ros-**version_name**-image-proc.

amakil76, 2024-03-25:
I'm having the same problems as the top first questions. The pi is using the image provide by your website (Ros kinetic  for Pi 3b+). Is this because the Kinetic is no longer supported ?

pi@raspberrypi:~ $ sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-pro
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-image-transport
E: Unable to locate package ros-kinetic-cv-bridge
E: Unable to locate package ros-kinetic-vision-opencv
E: Unable to locate package ros-kinetic-image-pro
pi@raspberrypi:~ $