ROBOTIS-GIT/emanual #663: Suggestions to open_manipulator documentation

Affonso-Gui, 2019-01-24:
Hello,

I am writing regarding some problems/suggestions to the open_manipulator documentation, described below:

- On section [4.4 Usb Settings](https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/platform/openmanipulator.md#usb-settings), we have
> Please run roscore before rosrun, because rosrun can't operate without roscore. The rosrun and roscore should be run in each other terminal.

This information is not correct. Rosrun can also be used as a simple way to run executables in a ros package. In this case, since `create_udev_rules` is a simple bash script without ROS dependencies, it is fine to rosrun it without a roscore (it does need to have /opt/<distro>/setup.bash sourced)

- On section [5. Run roscore](https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/platform/openmanipulator.md#run-roscore), it is not necessary to run the roscore before the following roslaunch.

ROBOTIS-Will, 2019-01-24:
Hi @Affonso-Gui 
Thank you for the feedback :)

routiful, 2019-01-25:
@Affonso-Gui 

Thanks. I deleted information about roscore in Usb Settings.
I know that running roscore before roslaunch commands is not necessary however to prevent accident when the launch file included roscore is broken I recommend that run roscore seperately.

closed 2019-01-25 by ROBOTIS-Will