ROBOTIS-GIT/emanual #729: nav2_util build error in ros2 tutorial

eavelardev, 2019-03-13:
ros2 tutorial error
```
colcon build --symlink-install
```
--- stderr: nav2_util                                                                                                                                                               
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp: In function ‘rclcpp::Node::SharedPtr nav2_util::generate_internal_node(const string&)’:
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: error: ‘NodeOptions’ is not a member of ‘rclcpp’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: note: suggested alternative: ‘InitOptions’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
           InitOptions
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:69:3: error: ‘options’ was not declared in this scope
   options.use_global_arguments(false);
   ^~~~~~~
make[2]: *** [CMakeFiles/nav2_util_core.dir/src/node_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_util_core.dir/all] Error 2
make: *** [all] Error 2

######

I don't know if I need to install a dependency, the error don't mention it

routiful, 2019-03-13:
ros2.repos has release-latest tag are not support NodeOptions.
Please use master branch

https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos

eavelardev, 2019-03-13:
With the master branch nav2_util was compiled well, but now I get another error:

~/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:32:5:   required from here
/usr/include/c++/7/ext/new_allocator.h:136:4: error: no matching function for call to ‘rclcpp::Node::Node(const char [11], const char [10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char> >&, std::vector<rclcpp::Parameter>&)’
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ~/ros2_ws/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28:0,
                 from ~/ros2_ws/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from ~/ros2_ws/install/rclcpp/include/rclcpp/rclcpp.hpp:144,
                 from ~/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:7:
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:604:3: note: candidate: rclcpp::Node::Node(const rclcpp::Node&, const string&)
   Node(
   ^~~~
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:604:3: note:   candidate expects 2 arguments, 5 provided
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:90:12: note: candidate: rclcpp::Node::Node(const string&, const string&, const rclcpp::NodeOptions&)
   explicit Node(
            ^~~~
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:90:12: note:   candidate expects 3 arguments, 5 provided
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:79:12: note: candidate: rclcpp::Node::Node(const string&, const rclcpp::NodeOptions&)
   explicit Node(
            ^~~~
~/ros2_ws/install/rclcpp/include/rclcpp/node.hpp:79:12: note:   candidate expects 2 arguments, 5 provided
make[2]: *** [CMakeFiles/image_transport-remapping.dir/test/test_remapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_transport-remapping.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< image_transport	[ Exited with code 2 ]

Maybe not all packages are updated

Yeshasvitvs, 2019-03-31:
@eavelardev I am having the same errors with `image_transport`. Did you manage to fix your installation ?