ROBOTIS-GIT/emanual #785: XM430-W210 : Drive Mode description

rkuroi, 2019-05-01:
Hi, as I used the XM430-W210 motors for a school project I noticed that time-based profile cannot be set unless firmware version is v42, and not v38 as the e-Manual states.
That being said, thanks for the Dynamixel Wizard that helped fix my problem !

ROBOTIS-Will, 2019-05-07:
@rkuroi Hi! Thank you so much for pointing that out. We'll revise the manual. Have a great day and enjoy your project :D

ROBOTIS-David, 2019-05-08:
@rkuroi  Hi. I am glad to hear that you notice there is an error of manual. Thank you ! 

As you recognize it, Some X-Series including "XM430-W210" requires firmware update "v42" to use time - based profile, not "v38" which is older version of firmware. Thanks to you, we worked it out and the contents are improved !  

You can check out modifications below link.
[Link1](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#drive-mode10)

[Link2](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#profile-acceleration108)

[Link3](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#profile-velocity112)

closed 2019-05-08 by ROBOTIS-Will