ROBOTIS-GIT/emanual #793: 9.1.3.3 Needs a roslaunch

Lloydf, 2019-05-11:
And Open the other terminal window and enter the following command in the terminal. This command is to execute a package that recognizes the ar marker. Enter the type of camera you are using and the size of the ar marker. In this example, we use a Raspberry Pi Camera V2 and 3cm ar marker.

$ open_manipulator_ar_markers ar_pose.launch camera_model:=raspicam user_marker_size:=3.0
(NOTE the roslaunch command is missing)
I.E $ roslaunch open_manipulator_ar_markers ar_pose.launch camera_model:=realsense_d435 user_marker_size:=3.0

ROBOTIS-Will, 2019-05-13:
@Lloydf 
Thank you for the feedback! :D

yhna, 2019-05-13:
@Lloydf 
Thank you for your contribution.
We will correct this issue as soon as possible.

closed 2019-06-11 by ROBOTIS-Will