ROBOTIS-GIT/emanual #1036: Raspian image checksum

mikegdodson, 2020-01-08:
I just downloaded the Raspbian image from the following link in the setup wiki:

http://www.robotis.com/service/download.php?no=1738

which redirects to an https site here:

https://www.robotis.com/service/download.php?no=1738

The sha256 checksum doesn't match the wiki.

I perform `shasum -a 256 image_rpi_20190429.img.zip` and get the following:

eb8173f3727db08087990b2c4e2bb211e70bd54644644834771fc8b971856b97

The wiki states the sum should be:

7a868c275169b1f02c04617cc0cce9654fd8222623c78b22d0a27c73a9609398

Has the image changed?

mikegdodson, 2020-01-08:
Update:  The sha256 checksum on the wiki is for the image, not for the downloaded zip file.

The instructions on both the wiki and the linked RPi Easy SD Card Setup instructions direct performance of the checksum before unzipping.

Unless I misunderstand something here, perhaps the wiki should be updated with the checksum of the zip file, not the image?

ROBOTIS-David, 2020-01-14:
@rjshim 
Please check the issue above.
If I need to edit e-Manual regarding the issue; SHA256, please let me know. 
Thank you

ROBOTIS-David, 2020-02-24:
@ROBOTIS-Will 

Please see sha256 checksum issue (Provided information from e-Manual and Wiki does not match.)

ROBOTIS-Will, 2020-02-24:
@ROBOTIS-David 
Please revise the eManual so that both zip and img checksum can be confirmed by users.

ROBOTIS-David, 2020-03-06:
@mikegdodson 

Hi, Both zip and image file of SHA 256 is now updated at [e-Manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#remote-pc)

Thank you for the report.

Have a great day.

closed 2020-03-06 by ROBOTIS-David