ROBOTIS-GIT/emanual #1079: Unbale to locate package when installed Additional Dependent Packages for AutoRace on Pi

Indruino-hieu, 2020-03-30:
Hi,

When I installed Additional Dependent Packages for AutoRace on Pi, it has the following issues . Could you please advise how can I fix it?

pi@raspberrypi:~$ sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-proc
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-kinetic-image-transport
E: Unable to locate package ros-kinetic-cv-bridge
E: Unable to locate package ros-kinetic-vision-opencv
E: Unable to locate package ros-kinetic-image-proc

Indruino-hieu, 2020-03-30:
Hi,Gilbert!
Could you please help? Thank you in advance!!
Regards,
Hieu.

ROBOTIS-Will, 2020-06-24:
@Indruino-hieu 
Sorry about the late reply. Required packages for Autorace are already installed in the provided Raspbian image, so please ignore those errors as they are not supporting binary installation.
Thanks.

ROBOTIS-Will, 2020-09-16:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen it when needed.

closed 2020-09-16 by ROBOTIS-Will