ROBOTIS-GIT/emanual #1154: Turtlebot3 friends's BOM is wrong

DaisukeUra, 2020-07-03:
Hi,

I want to build Turtlebot3 Mecanum(https://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-mecanum).
So, I checked BOM and bought mecanum wheels rb-nex-13 and printed joint parts.
(BOM: https://docs.google.com/spreadsheets/d/1vbIFMRJMbyOd2D6BtwV1Ow_yBc9xRHktqZhTTmwI_l4/edit#gid=953634249)

But they can't connect because of hole placement is wrong.
I think the link address of wheels in BOM is wrong.

Thanks.

ROBOTIS-David, 2020-07-10:
@DaisukeUra 

Thank you for reporting the issue. 

Sorry about the inconvenience that may have caused and we are checking the BOM now, 

If you do not mind, could you specify the name of item from the item list below? 

https://us.misumi-ec.com/vona2/detail/110302280810/?HissuCode=NKJ3-8&PNSearch=NKJ3-8&KWSearch=NKJ3-8&searchFlow=results2type
--
https://us.misumi-ec.com/vona2/detail/221000551286/?HissuCode=CSH-ST-M2.5-8&PNSearch=CSH-ST-M2.5-8&KWSearch=CSH-ST-M2.5-8&searchFlow=results2type
https://us.misumi-ec.com/vona2/detail/221000551286/?HissuCode=CSH-ST-M2.5-12&PNSearch=CSH-ST-M2.5-12&KWSearch=CSH-ST-M2.5-12&searchFlow=results2type
https://us.misumi-ec.com/vona2/detail/221000551286/?HissuCode=CSH-ST-M3-40&PNSearch=CSH-ST-M3-40&KWSearch=CSH-ST-M3-40&searchFlow=results2type
https://us.misumi-ec.com/vona2/detail/110300250540/?HissuCode=SLBNR3&PNSearch=SLBNR3&KWSearch=SLBNR3&searchFlow=results2type
https://us.misumi-ec.com/vona2/detail/110500137950/?HissuCode=SB-O3-45&searchFlow=results2similartn
https://www.amazon.com/100Mm-Aluminum-Mecanum-Wheel-Right/dp/B01CTUT4GY/ref=sr_1_fkmrnull_1?keywords=100mm+Mecanum+Wheel&qid=1549936037&s=gateway&sr=8-1-fkmrnull
https://cad.onshape.com/documents/1131a9f555bd1842665c39d0/w/1ed900186382ba9b6a9c800b/e/a62bcab3415cef911d0ba0bf

Thank you.

DaisukeUra, 2020-07-10:
@ROBOTIS-David 
Thank you for your response.

I think `Mecanum wheel` in that list is wrong.
The link shows that the mecanum wheels the diameter is 100mm.
However, the `3D printing parts(Wheel joint)` link shows that the wheels are 60mm mecanum wheels.
(Unfortunately, I don't know the model number of the wheel used in this CAD model.)

Thank you.

---
P.S. 
I designed a new wheel joint for 100mm wheels, and mecanum turtle now moves.
Anyway, the Turtlebot 3, especially the modular plates and Dynamixel Servo Motor, are great! thank you.

ROBOTIS-David, 2020-07-10:
@DaisukeUra 
I see, 

Thank you for your attention to the TurtleBot friends. 

@ashekim 

Ashe, Could you please see the [3D printing parts(Wheel joint)](https://cad.onshape.com/documents/1131a9f555bd1842665c39d0/w/1ed900186382ba9b6a9c800b/e/a62bcab3415cef911d0ba0bf) and check out the diameter of the part. One of ROBOTIS user has reported that wheels of the parts has wrong measured size 60 mm which does not fit in the [Mecanum wheel](https://www.amazon.com/100Mm-Aluminum-Mecanum-Wheel-Right/dp/B01CTUT4GY/ref=sr_1_fkmrnull_1?keywords=100mm+Mecanum+Wheel&qid=1549936037&s=gateway&sr=8-1-fkmrnull), 100 mm. 

Thank you,

ashekim, 2020-07-13:
@ROBOTIS-David 

David, I think we can modify the BOM with the link below.

[https://www.amazon.com/Mecanum-UniHobby-Compatible-Raspberry-Arduino/dp/B07CK84N69/ref=sr_1_18?dchild=1&keywords=60mm+Mecanum+Wheel&qid=1594607324&sr=8-18 ](url)

Thank you.

ROBOTIS-David, 2020-08-25:
@DaisukeUra 

Hi, the issue is resolved, 

I would like to close this issue if you have no additional inquiry ! 

Thank you :)

closed 2020-08-25 by ROBOTIS-David