ROBOTIS-GIT/emanual #1162: Profile Acceleration and PWM

swinterbotix, 2020-07-16:
If a motor is set to operate in PWM mode, does the Profile Acceleration register do anything? From the docs, it says that Profile Acceleration applies to all control modes except for current - but I fail to see how it applies to PWM as well...

ROBOTIS-David, 2020-08-11:
Hi !@swinterbotix  

I am very sorry for missing your ticket and thank you so much for your report. 

Yes you are right, PWM Mode is not compatible with Profile Acceleration as it controls the duty ratio ! 

The contents has been updated, please [check it out](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-acceleration108) !

swinterbotix, 2020-08-11:
Awesome! Looks good. Thanks!

closed 2020-08-11 by swinterbotix