ROBOTIS-GIT/emanual #1197: Specify hardware setup for motor setup

agutenkunst, 2020-09-06:
Reading https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3
I think it would be useful to be more specific on the hardware setup. 
Reasonably the USB needs to plugged into the PC, but at which step do I need to connect the OPENCR 
to power supply?

 I figure having connected the power supply all the time should do no harm, yet some more specification could be useful to people getting started on the topic.

ROBOTIS-David, 2020-09-07:
@agutenkunst 

Hi, agutenkunst, thank you for your valuable feedback. 

You are definitely right. As you mentioned, The current setup instruction could make user confused when they reading. 

I will make it clearer with adding H/W step ! 

Thank you so much !!

ROBOTIS-David, 2020-09-07:
@ashekim 
@ROBOTIS-Will 

Hi, guys. 

I received valuable feedback from one of the TB3 user, please refer to it and give me idea. 

Or I just add "1. While the power is being supplied to the board, Click on the `Reset` button with the `sw` button being pressed on of OpenCR 1.0. 

Thank you,

ROBOTIS-Will, 2020-09-07:
I'd simply mention to connect both USB and power before the `1. Upload Setup Firmware to OpenCR`.

ROBOTIS-David, 2020-09-08:
@ROBOTIS-Will 

Thank you for your suggestion, I think the suggested sentence by you will be clear enough :) 

I would like to close this issue ticket if there is no more progress.

closed 2020-09-08 by ROBOTIS-David