ROBOTIS-GIT/emanual #1226: Unable to get lidar data in ROS2

sjev, 2020-11-09:
After reading the documentation and hours of googling through issues ([like this one](https://github.com/ros2/ros2_documentation/issues/569)), I'm still not able to read the lidar data published on `/scan` topic.

How to reproduce:
1. bring up tutrtlebot
2. start `rqt` on remote pc. -> there are no updates from the scan sensor (below is a screenshot from the emanual, showing *unknown* update frequency for the data. Nothing is coming in)
[image]
3. check topic with `ros2 topic echo /scan` -> **messages are being received**
4. subscribe from python `sub = node.create_subscription(
        String, 'scan', chatter_callback, qos_profile=qos_profile_sensor_data)` -> **no messages**

I'm at a loss here...

sjev, 2020-11-09:
Ok, I did manage to get the data with this code, but the documentation could be a bit better.

``` python
  1 from rclpy.qos import qos_profile_sensor_data, QoSProfile
  2 from sensor_msgs.msg import LaserScan
  3 import rclpy
  4 
  5 def chatter_callback(msg):
  6     print(msg)
  7 
  8 def main():
  9     
 10     rclpy.init()
 11     qos = QoSProfile(depth=10)
 12     node = rclpy.create_node('scan_listener')
 13     sub = node.create_subscription(LaserScan,
 14                                    'scan',
 15                                    chatter_callback,
 16                                    qos_profile=qos_profile_sensor_data)
 17     try:
 18         while True:
 19             rclpy.spin_once(node)
 20 
 21     except KeyboardInterrupt:
 22         pass
 23 
 24 if __name__ == "__main__":
 25     print('Starting scan listener')
 26     main()
 27     print('done.')        
```

ROBOTIS-David, 2020-11-10:
Hi, Thank you for reporting the issue :) 

I would like to transfer this issue to the TurtleBot3 issue section. 

One of our team member will address your issue as soons as possible.

Thank you,