ROBOTIS-GIT/emanual #1239: add "curl" to instructions @ 3.2.5 Install ROS Noetic Ninjemys

MikeHallettUK, 2020-12-07:
Suggest change to add "curl" to instructions @ https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/

3.2.5 Install ROS Noetic Ninjemys

change = 

$ https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_noetic_rp3.sh

to = 

$ curl https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_noetic_rp3.sh --output install_ros_noetic_rp3.sh

?

MikeHallettUK, 2020-12-07:
Also @ 2. Install dependencies using rosdep.

change = 

$ cd ~/colcon_ws && rosdep update && rosdep install --from-paths src --ignore-src -r -y

to = 

$ cd ~/catkin_ws && rosdep update && rosdep install --from-paths src --ignore-src -r -y

MikeHallettUK, 2020-12-07:
Also @ 3. USB Port Setting - I needed to do "sudo apt install ros-noetic-rosbash" before running:

$ rosrun turtlebot3_bringup create_udev_rules

ROBOTIS-Will, 2020-12-09:
Thank you for reporting the errors.

> Also @ 3. USB Port Setting - I needed to do "sudo apt install ros-noetic-rosbash" before running:
> 
> $ rosrun turtlebot3_bringup create_udev_rules

When installing ros-noetic-ros-base, ros-noetic-rosbash should have been installed. Let me double check if this is missing in the ros base package.

MikeHallettUK, 2020-12-12:
Closed: no further issue with me.

Although it may be useful to suggest upgrading to 1.4.16 OpenCR Arduino code line if was already installed for e.g. Melodic.

closed 2020-12-12 by MikeHallettUK