ROBOTIS-GIT/emanual #1289: Protocol 1.0 Where are the parameters

HarbourXbr, 2021-02-01:
I read Protocol 1.0，but i can't see the parameters. How can I get these parameters .

ROBOTIS-David, 2021-02-01:
@HarbourXbr Hi, Could you please explain little bit more about your questions? It would be great If I know what product you are using ( I guess DYNAMIXEL) and what you are planning to do with the product. Thanks !

HarbourXbr, 2021-02-02:
@ROBOTIS-David  Hi, like this one, 
[image]

Sync Write the paramter1 starting address, Where can I get a detailed description of these addresses

HarbourXbr, 2021-02-02:
@ROBOTIS-David I am going to use my own MCU to communicate MX28

ROBOTIS-David, 2021-02-02:
@HarbourXbr 

The Parameters are the [Instruction](https://emanual.robotis.com/docs/en/dxl/protocol1/#instruction)’s auxiliary data field.

Using these parameters combining the Instruction, you can access your DYNAMIXEL's control table (firmware address) to read/write data. 

Basically, the packet construction as follows, 

Header1 | Header2 | Packet ID | Length | Instruction | Param 1 | … | Param N | Checksum
-- | -- | -- | -- | -- | -- | -- | -- | --
0xFF | 0xFF | Packet ID | Length | Instruction | Param 1 | … | Param N | CHKSUM

You can simply test using the DYNAMIXEL Wizard 2.0's packet communication. 

[image]

As you can see, I sent "Write" Instruction (Enabling Torque) to the DYNAMIXEL, and the packet construction is as below. 

Header1 | Header2 | Packet ID | Length | Instruction | Param 1 | Param 2 | Checksum
:--: | :--: | :--: | :--: | :--: | :--: | :--: | :--:
| FF | FF | 01 | 04 | 03 | 18 | 01  | DE |

**Note that the numbers are Hex**

- Param 1 represents Starting Address of the Torque Enable (Address 24, Dec)
- Param 2 represents data you want to write to the address. 

To know the Address, see your product's control table items in [e-Manual](https://emanual.robotis.com/docs/en/dxl/mx/mx-28/#control-table-description)

Note: If you would like to develop a system using DYNAMIXEL, use [DYNAMIXEL SDK](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/overview/). It supports various programming language C used as binding. From the given link (e-Manual), select the programming language that you plan to use. 

For more inquiries, you can email me via support@robotis.com

HarbourXbr, 2021-02-02:
@ROBOTIS-David So the address has nothing to do with the communication protocol , the MX28T product can be used differently
  protocol，potocol 1.0 or 2.0? I just need to look at the product‘s control table items?

ROBOTIS-David, 2021-02-16:
@HarbourXbr 

I hope that it's not too late. 

1. The DYNAMIXEL Protocol 1.0 and Protocol 2.0 basically has a different control table. 
2. MX(2.0) is a special firmware for the DYNAMIXEL MX series supporting the DYNAMIXEL Protocol 2.0. The MX(2.0) firmware can be upgraded from the Protocol 1.0 by using the Firmware Recovery in DYNAMIXEL Wizard 2.0. 

Thank you,

closed 2021-03-29 by HarbourXbr