ROBOTIS-GIT/emanual #1363: Task3.0 support for tablets

BManCan, 2021-05-18:
Any update on resolving the resolution / compatibility for the Task 3.0 app on iOS tablets?
Thanks.

ROBOTIS-David, 2021-05-20:
@BManCan 

Hi BMan, 

Firstly, I am very sorry for your inconvenience, 

Unfortunately, the resolution issue has not been resolved. 

Until resolved, using R+Task 3.0 on Windows/Mac system is highly recommended for use of all features in R+Task 3.0.

If you have more questions, please reach out support@robotis.com.  

Thank you,

ROBOTIS-Yang, 2025-01-08:
@BManCan

Are there any further issues you've encountered related to this matter?
We're actively working on updates for R+Task 3.0. Should the problem persist, please don't hesitate to reopen the issue.

closed 2025-01-08 by ROBOTIS-Yang