ROBOTIS-GIT/emanual #1420: Dynamixel SDK and CM-530

Crypt-Cesar-Labs, 2021-10-06:
Hello, I am working on a project with a Bioloid robot and AX-12A Servos. I just want to ask if it is possible to use the Dynamixel SDK and the CM-530 as an intermediary to control the servos. I only have the CM-530 and Linux, so the kernel detects the CM-530 serial protocol without problem, so I want to ask if I have the opportunity to use Dynamixel on the CM-530, because roboplus does not run natively on linux.

ROBOTIS-David, 2021-10-08:
Hello, @Crypt-Cesar-Labs 

From my understanding, you are questioning how to implement CM-530 as data converter to exchange packets between PC and DYNAMIXEL. 

Unfortunately, the approach may not be achievable. If you would like to use popular code language technique, take [Embedded-C ](https://emanual.robotis.com/docs/en/software/embedded_sdk/embedded_c_cm530/#programming)into consideration.

ROBOTIS-Yang, 2025-01-08:
This issue has been inactive for a long time. We're closing it now. Please reopen if necessary

closed 2025-01-08 by ROBOTIS-Yang