ROBOTIS-GIT/emanual #1451: [TurtleBot3] Simulations may be missing configuration of `GAZEBO_PLUGIN_PATH`

goekce, 2022-01-22:
I use Galactic and compiled from source on Archlinux.

I wanted to start the following simulation according to [Gazebo-simulation](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#gazebo-simulation):
```
$ export TURTLEBOT3_MODEL=waffle_pi
$ ros2 launch turtlebot3_gazebo empty_world.launch.py
```
I got the following error:
```
...
[gzserver-1] malloc(): corrupted top size
[ERROR] [gzserver-1]: process has died [pid 4177, exit code -6, cmd 'gzserver /opt/ros/galactic-turtlebot3-quickstart/share/turtlebot3_gazebo/worlds/empty_world.world                                                                      -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[ERROR] [gzclient   -2]: process has died [pid 4179, exit code 255, cmd 'gzclient   '].
...
```

The solution was to append the path where `libgazebo_ros_init.so` can be found to `GAZEBO_PLUGIN_PATH`. For example:

```
$ find /opt/ros -iname libgazebo_ros_init.so
/opt/ros/galactic-turtlebot3-quickstart/lib/libgazebo_ros_init.so
$ GAZEBO_PLUGIN_PATH+=/opt/ros/galactic-turtlebot3-quickstart/lib:
$ ros2 launch turtlebot3_gazebo empty_world.launch.py
```
I am not on a supported platform and I do not know if this problem also occurs in Ubuntu. I could only find an older issue: https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/43

This problem does not happen if I start another simulations like:

```
gazebo --verbose <gazebo-ros-pkgs-path>/share/gazebo_plugins/worlds/gazebo_ros_diff_drive_demo.world
```
It would be great if this notice could be included in the docs.

ROBOTIS-Yang, 2025-01-08:
Hi, goekce

ROBOTIS-Yang, 2025-01-08:
Hi, goekce

Given the extended period of inactivity, we're marking this issue as closed. If you believe this issue is still relevant, please let us know and we'll be happy to reopen it.

closed 2025-01-08 by ROBOTIS-Yang