ROBOTIS-GIT/emanual #1479: in Matlab the groupSyncWrite can't be find

chenxinencode, 2022-04-26:
when I run the program,It went wrong in code"group_num = groupSyncWrite(port_num, PROTOCOL_VERSION, ADDR_MX_GOAL_POSITION, LEN_POSITION);",
matlab show that it can't find the lib"groupSyncWrite" when execute the code:VALUE = calllib(lib_name, 'groupSyncWrite', group_num, protocol_version, start_address, data_length);
what should I do?

ROBOTIS-David, 2022-04-26:
Unfortuanately, the issue section on the emanual is not proper channel. 

I highly recommend visiting the [community](https://community.robotis.us/c/community/5) page, and discuss your issue with users. (Our technical experts are also actively playing around there)

ROBOTIS-Yang, 2025-01-08:
This issue has been inactive for a long time. We're closing it now. Please reopen if necessary.

closed 2025-01-08 by ROBOTIS-Yang