ROBOTIS-GIT/emanual #1565: [Improvement] Please indicate that LDS_MODEL variable should be set on both PC and TurtleBot3

kotokaze, 2023-05-26:
I recently moved to ROS 2.
I get an error and abort when trying to use LiDAR with turtlebot3_cartographer and others with old TurtleBot3s. However the newer one with LDS-02 works as expected.

Finally, I noticed that the cartographer seems to run with LDS-02 by default. It took me a long time to realize that the environment variable `LDS_MODEL=LDS-01` needs to be set on the PC side as well as on the bot itself.

Could you please note to set LDS_MODEL at least when using LDS-01 in the documentation?

https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#new-lds-02-configuration-4

shiba-8ro, 2023-05-28:
Upvote +1 .

GyuH13, 2025-02-24:
Hi @kotokaze 
I apologize for the late reply. The E-manual has been updated with information regarding the LDS environment variables. Thank you for your advice. I will consider this issue closed.

closed 2025-02-24 by GyuH13