ROBOTIS-GIT/emanual #1667: Item Size

mukyokyo, 2024-04-23:
The Size (Byte) listed in the Y series control table is inaccurate across the board.
It has been several weeks since I pointed out that problem to a certain place, but it has not been corrected.

mukyokyo, 2024-06-14:
It is inherently wrong that the person pointing out the problem is an outsider.
I am not considering closing it, as it is for you to verify that the correction made by the earlier point is complete.

In conjunction with this matter, I would like to point out that the handling of 32bit numbers is not correct.
It should also be noted that there are differences between the behavior of the actual device and the description in the e-manual.

ROBOTIS-Yang, 2025-01-08:
We apologize for any inconvenience this may have caused. The issue has been resolved. Thank you for your understanding.

closed 2025-01-08 by ROBOTIS-Yang