ROBOTIS-GIT/emanual #1702: Humble tutorial and SSH

haavarpb, 2024-09-23:
Hello,

I am in the process of setting up several turtlebot3's for educational purposes. We are using ROS2 Humble.

The tutorials assume that you can SSH into the robot knowing its IP. However, you can verify that the Humble tutorial is the first to use a generic Ubuntu Server 22 LTS image instead of the custom image provided in the other distros' tutorials.

This generic ubuntu server image uses cloud-init to set up the SSH server daemon to refuse password authentication. This can be verified by trying to ssh into a fresh ubuntu server instance.

I will try to see how I can change this behaviour myself, but I suggest you update the tutorials either by showing how to ssh into the turtlebot3 or provide a customized ubuntu server image with a cloud-init configuration that suits educational requirements.

The specific error that I get when trying to `ssh user@turtlebot_ip` is `pubkey denied`. I found that by modifying `/etc/ssh/sshd_config` I can uncomment line 57 `PasswordAuthentication yes` and changing line 62 `KbdInteractiveAuthentication yes`.

The ssh client was able to connect to the server by (I can't remember the exact, but the point is that you need to specify this in the client)

`ssh -o PreferredAuthentication=keyboard-interactive user@turtlebot_ip`

I guess the ideal state for such an educational system would be to allow entering the password the first time, and then allowing copying over public ssh keys using something like `ssh-copy-id`.

Atleast that's what I was thinking. Thoughts?

GyuH13, 2025-02-24:
Hi @haavarpb 
I apologize for the late reply. 
Thank you for your advice. As your opinion, i have posted the SSH connection-related settings in the E-manual.
I will update the manual in the future to include SSH setup when installing the OS on the SD card using RPI IMAGER.
Thank you.

closed 2025-02-24 by GyuH13

haavarpb, 2025-02-24:
Hi, and thanks for coming back to me. Do you happen to have a link to a PR or similar for me to check it out?

GyuH13, 2025-02-24:
Hi @haavarpb 

[image]
You can find the relevant section in the SBC setup part of the E-manual. If you encounter any inconveniences while using the manual, please provide feedback, and I will review it and make improvements when possible.
Thank you.

haavarpb, 2025-02-24:
Awesome, thanks :-)