ROBOTIS-GIT/emanual #1726: LDS-01 dimensions

pitosalas, 2024-12-21:
Where can I find an accurate dimensional drawing for the four holes of the LDS-01 lidar? I've looked around and can't find it!

shiba-8ro, 2024-12-22:
@pitosalas You can check the dimention on OnShape that Turtlebot3 official hw document.
https://cad.onshape.com/documents/2586c4659ef3e7078e91168b/w/14abf4cb615429a14a2732cc/e/9ae9841864e78c02c4966c5e

robotpilot, 2024-12-22:
Hi, @pitosalas, You can find detail specification document here: 
https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_lds_01/#detail-specification-document
Thanks, @shiba-8ro :)

robotpilot, 2025-02-24:
This issue has been open for a long time, so I will close it. Please reopen if necessary.

closed 2025-02-24 by robotpilot