ROBOTIS-GIT/emanual #1731: [Humble] SBC Setup build of turtlebot3 frezes on turtlebot3_node build (possibly memory issue)

LUGGGI, 2025-01-06:
Hello everyone,

I noticed that the build of `humble https://github.com/ROBOTIS-GIT/turtlebot3.git` gets stuck on the turtlebot3_node package when building on a raspberry pi with only 2GB of memory (with swap disabled (I think this is the default on ubuntu server 22.04.5 (64 bit) when installed with raspberry pi imager)).
Htop also shows full memory usage at this point.

Fix:

- Building on Pi with more than 2GB memory
- Maybe enable swap, but I haven't tried that.

GyuH13, 2025-01-09:
Hi @LUGGGI 
We recommend enabling swap memory to optimize performance in low-memory environments. Additionally, please ensure the following:

- Have you installed the Ubuntu Server OS on your Raspberry Pi?
- Have you removed Cartographer and Navigation2 from the TurtleBot3 workspace?
- Are there any other unnecessary processes running on the system?  

If possible, could you provide additional information, such as the output of $ free -h to check your RAM usage? This will help us assist you more effectively.

LUGGGI, 2025-01-09:
Hi @GyuH13 ,
thanks for the reply. 

- Yes, I have installed Ubuntu Server
- Yes, I removed it (by following the setup tutorial)
- It was a completely fresh install of Ubuntu Server, so I don't think so.

I was able to get it to work using a different pi (with more memory), so it is not really practical to try to reinstall it now. But if I experience it again I will post the details of the RAM usage or maybe others have the same problem.

GyuH13, 2025-01-13:
Hi @LUGGGI 
I tested the build on a 2GB Raspberry Pi under similar conditions and observed that the build process stopped unexpectedly. To resolve this, I created a 2048MB swap file and re-ran the build, which completed successfully in approximately 5 minutes.
As a result of checking the RAM usage on my Raspberry Pi, I found that using swap is essential for building.

If you have any additional questions or need further assistance, please feel free to reach out.  
This issue will closed in one day.

closed 2025-01-14 by GyuH13